bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz #2: What do I connect to vref?

delchrys, 2017-04-17:
In the schematics vref is named. But what should I connect it to?
And do you have some better pictures of your breadboard setup or a parts list from Farnell or something?

bbqkees, 2017-04-17:
You only need to connect every line that is named U_Ref. No exteral connections.
It is an internal reference voltage.

For a picture of the breadboard see the Documentation page again, I added one.

I do not have a parts list, there are only a few parts in the schematic, it is not that difficult.

delchrys, 2017-04-17:
So the u refs are connected to each other or to the lm393?

bbqkees, 2017-04-17:
Draw a line from each U_Ref to the other and remove the label, maybe that will make it more clear.
The labels are there just so do not have to show crossing wires.

delchrys, 2017-04-17:
Ah ok I get it. Now it's clear for me thanks. Will buildo it soon to test if it works.

bbqkees, 2017-04-17:
So now its clear I will close the issue.

closed 2017-04-17 by bbqkees

delchrys, 2017-04-18:
Sorry bbqkees i have another question. I see you used all through hole components. Which diodes (schottky) you used instead of a smd Bat54s+ and BAT46 package??

bbqkees, 2017-04-18:
You can probably use 1N4148 diodes for all.
I used two different diode types (1N4148 and the other of unknown type but identical or similar), but only because I had those at hand.

delchrys, 2017-05-08:
another question,
i'm not that electric but i think i can make it work, but does the lm393 also need vcc and ground connected?? on pin 8 and 4???

bbqkees, 2017-05-08:
Yes you need to connect those pins too, otherwise the chip itself has no power.

delchrys, 2017-05-08:
thanks will do that, stupid mistake

delchrys, 2017-05-08:
Yes it works now. Thanks for the help.
Is it possible to get more data from the EMS bus??? In the documentation i read a lot of other info, but how can i find out if this is all the data my Nefit spits out?

delchrys, 2017-05-10:
question, how did you figure out the addresses, can't seem to find any similar entry's in the wiki.
for instance 0x08,0x18,0x12,0x02,

bbqkees, 2017-05-11:
So if you read this page VERY carefully:
https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme
You see that the offset for the first databyte is 5. So the first databyte is at the 5th byte of the message.

Now check the UBAMonitorFast table.
You can see f.i. 08 | 00 | 18 | 23 | 1 | ASCII | Service-Code 1. Zeichen
Sender: 0x08
Type: 0x18
Byte offset from start: 23 DEC = 0x17
Description: Service Code 1st sign.

Now note the sequence of your question: 0x08 | 0x18 | 0x12 | 0x02 | //#9   8 status code 1st letter.
Sender: 0x08
Type: 0x18
Byte offset from first databyte: 0x12 (remember it is +5 bytes from the start so 0x12+5=0x17)
Datatype: 0x02 = ASCII (Only used in the sketch) 
Description: //#9   8 status code 1st letter

So if you check the other parameters in the sketch you see that most of them are just a decoding of the UBAMonitorFast and UBAMonitorWWMessage message.