bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz #5: Link to schema is dead.

jrswgtr, 2017-11-20:
Hi there. I like your project and like to use parts of it for myself. I clicked the link for the breadboard schema in your documentation but it seams to be (re)moved. Do you by any chance still have the schema yourself?

jrswgtr, 2017-11-20:
I am mainly interested in the Arduino and breadboard part. I want to do other stuff with the data. Is it only possible to read values, or can you also set values to the boiler?

bbqkees, 2017-11-21:
Have you checked the [Documentation](https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/tree/master/Documentation)?

The 'dead' reference is the page where I originally got the circuit schematic from, but it is already displayed at the top of the documentation page. There is no separate breadboard schematic.

If you create the top half of the circuit you can only read from the bus.
If you create the whole circuit you can also write to the bus.

closed 2018-01-15 by bbqkees