bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz #6: conflict_with_netifserial_and_sd

jmmprodukts, 2017-12-27:
Ik wil de nefitserial SD library samen in een sketch toepassen. echter ontstaat tijdens compilatie een 'conflict'met HardwareSerial0.cpp. Daar een beginnende Arduino en C hobbyist ben, heb ik geen idee hoe ik dit oplossen kan.

[conflict_with_netifserial_and_sd.zip](https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/files/1589331/conflict_with_netifserial_and_sd.zip)

**Archiving built core (caching) in: C:\Users\Jacoba01\AppData\Local\Temp\arduino_cache_130884\core\core_arduino_avr_megaADK_5638c217b04eda8279099c4937887265.a
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':

(.text+0x0): multiple definition of `__vector_25'

libraries\Nefitserial\NefitSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

exit status 1
Fout bij het compileren voor board Arduino Mega ADK**

bbqkees, 2017-12-27:
Kan zijn dat je de NefitSerial library niet in de juiste library directory hebt geplaatst.

Probeer eerst eens of je een sketch kunt compilen met alleen NefitSerial.

(Verder moet je voor het lezenn van de SD kaart ook SPI.h includen.)

jmmprodukts, 2017-12-28:
Kijk even naar ingesloten screenshots, dan kan je zien dat alleen de includes al voldoende zijn om een conflict te realiseren.
[image]
[image]
[image]

bbqkees, 2017-12-28:
Ik zie het. Lijkt erop dat beide libraries dezelfde hardware resource proberen te defineren. Ik zal kijken of ik een aanpassing kan verzinnen.

jmmprodukts, 2017-12-29:
Na enig onderzoek gevonden dat SdFile.cpp gebruik maakt van bv 'Serial.print', hier lijken Serial en nefitserial te conflicteren. als ik nefitserial.h include en serial.print gebruik gaat het fout. Dit is normaal geen probleem daar hiervoor nefit.print in de plaats gebruikt kan worden. echter blijkt in 'SdFile.cpp' serial.print te worden gebruikt wat dus het conflict veroorzaak met nefitserial.h

[image]
[image]

bbqkees, 2018-01-02:
Je kunt proberen om de serial.print in SDFile enz te veranderen naar nefitserial.print.

bbqkees, 2018-01-09:
En al iets verder gekomen?

jmmprodukts, 2018-01-10:
Als ik alle "serial." in SD.h gerelateerde bestanden vervang door "//serial." (opmerking van gemaakt), dan klaagt de compiler niet meer.
Als ik alle "serial." in SD.h gerelateerde bestanden vervang door "nefitSerial.", dan klaagt de compiler op enkele plaatsen dat nefitSerial niet bestaat. dus in enkele .h en .cpp bestanden nefitSerial.h toegevoegd. nu werkt SD_nefit_SD.h reeds samen met mijn netwerk app plus sd reader. nu nog samen intergreren met de EMS monitor app.
[SD_nefit.zip](https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/files/1620141/SD_nefit.zip)

closed 2018-01-11 by bbqkees