bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz #7: support for ESP8266

mareksugar, 2018-03-16:
Hi bbqkees,

I am trying to compile program for Wemos D mini board, but it end with _'nefitSerial' was not declared in this scope_.
Could you point me out what need to be change in NefitSerial.h library to support ESP8266 boards?

bbqkees, 2018-03-16:
The Wemos D1 Mini is based on the ESP8266, which is incompatible with the NefitSerial library.
The NefitSerial library is meant for use with Atmel AVR type Arduino's like the Uno and Mega. 

Adaptation for the completely different chipset of the ESP8266 therefore needs considerable rework of the ESP Arduino library.

What you can do instead is use an Uno/Mega to interface with the EMS bus and the Wemos to transfer data to and from the Arduino with Wi-Fi.

bbqkees, 2018-03-16:
By the way the specific error you get usually means you did not include the library correctly.
So the library needs to be installed on your pc in the Arduino library directory. And afterwards you need to restart the Arduino IDE otherwise it can't see it.

But as just mentioned, you can't use this library with the ESP anyway.

mareksugar, 2018-03-16:
Yes, my fault, error which I have is _Error compiling for board WeMos D1 R2 & mini._

I mean I asked badly.
If library nefitSerial is just little bit modified library HardwareSerial, what you mentioned here:
https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/issues/3

could you help me/point me what needs to be change/add in library HardwareSerial.cpp for ESP8266
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/HardwareSerial.cpp
to modify it same way as you modified AVR HardwareSerial library.

bbqkees, 2018-03-16:
The basic thing to do is to bring up the ESP8266 interrupt for frame error into the library, so you can add functions that can act on the occurrence of the frame error.
If you check the readme page of the NefitSerial library there is some more info on that.

You need to modify the core libraries uart and HardwareSerial.

The second thing you need to do is to figure out how to generate a frame error sequence on the low level ESP UART.

Both are not that straightforward to do. The original additions to the Atmel-type Arduino serial library might indeed seem simple but they took considerable work and testing by the creator to find out.

mareksugar, 2018-03-19:
thank you, I didnt notice README under Nefitserial
I found post about serialEvent implementation in ESP8266
https://github.com/esp8266/Arduino/issues/2237
so I will try this way

closed 2018-03-19 by mareksugar

proddy, 2018-05-23:
@mareksugar check out https://github.com/proddy/EMS-ESP-Boiler. I worked with bbqkees on the implementation of his circuit for the ESP8266

mareksugar, 2018-05-23:
thank you so much, it looks really interesting :)