bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz #15: Connection lost with Wemos

rmarinusse, 2019-02-21:
Dear Kees,

I received your board + Wemos last week and got everything up and running relatively quickly using Domoticz/MQTT. I could see all data coming in, but after a while, usually a couple of hours, nothing was being sent until the Wemos was power cycled. After consulting with Proddy, he suggested to flash the board with his latest version of the sw to make sure that wasn't an issue. I did that, but now all I can get out of the board is this:

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

So, now my question to you is: How did you flash the Wemos firmware? What tools and settings did you use?

Thanks for your help

Ron

bbqkees, 2019-02-21:
Hi,
-This is not an issue regarding this Arduino repository.-

But anyway;
Which firmware was on the Wemos when you received it?
If it had a sticker on the ESD bag its 1.52 which is Proddy's latest version.
Otherwise it was 1.3x.

I grab the latest firmware (the bin file) from Proddy's repository and upload it to the Wemos with the ESPtool from Espressif.
https://github.com/espressif/esptool

bbqkees, 2019-02-22:
I'm going to send you another Wemos so I'll close this issue.

closed 2019-02-22 by bbqkees