bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz #16: How can I write values?

lobocobra, 2019-03-02:
I have a Buderus GB125 oil heating and have the EMS Bus interface board and the WEMOS.

**What I want to do**
I would like to change the temperatures on my Buderus GB125
=> Can I also write values to the EMS BUS and change for example the temperature settings?

**Hints needed**
Do I have to create a telegram ? How Can I send it?
https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme

Many thanks for some hints to be able to change settings.

lobocobra, 2019-03-03:
It seems that the data is written but my RC35 does not change thestatus.
Example:
If I set the heating manually to night mode, my rc35 keeps showing auto mode.
=> same if I set temperature.

Is this a problem with the RC35, that it does not listen to the BUS and thus the RC35 shows the old (overwritten) data?

Does anyone know this?

lobocobra, 2019-03-03:
A probable cause could be that my house has a floor-heating and the used circuit is NR2.
=> I currently try to send telegram but, I have problems to find plausible values.

If anyone  managed to change day/night or the temperature, please give me a hint...

lobocobra, 2019-03-04:
I found the solution...
The problem was that this code assumes that you have 1 circuit.
=> but if you have a floor heating it is circuit 2.

To change the day / night status you need to.... 
send 0b 90 47 07 01  => to show the current status
send 0b 10 47 07 02 => to set the status on automatic (2)

means you need to find out the codes... go here....
https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#schaltzeiten_allgemein

closed 2019-03-04 by lobocobra

proddy, 2019-03-04:
For the record, the code was designed that way and its clearly mention in the documenaton [here](https://github.com/proddy/EMS-ESP#which-thermostats-are-supported) _"RC35 with support for the 1st heating circuit (HC1)"_. Why don't you contribute and extend the code to support the other HCs?

lobocobra, 2019-03-04:
I am on it :)
=> Just wanted to leave a trace here for others so that they might use it.