bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz #19: Not function interface

kkberg, 2019-11-10:
I got an interface a few days agao, but it doesn't work. I tried a WEMOS D1 Mini and a PI 3b+. My heating system is with a Loagamax Plus GB172-14. I've also written 2 emails to you, but didn*t get any answer. EMS-ESP shows always that EMS-Bus is not connected. I tried the Rx resistor setting with the yellow jumper and the connection over jack or terminal. Would be nice to get some help to solve the problem. Bus voltage is 16v and it's the only thing I can measure with my multimeter. I don't have an ozciloscope. Thank you Klaus

bbqkees, 2019-11-11:
Try swapping TX and RX.

kkberg, 2019-11-11:
doesn't work either, also swapping jumper doesn't change anything

bbqkees, 2019-11-11:
Can you tell me exactly which pins on the board you connected to which pin on the Wemos?

kkberg, 2019-11-11:
first I tried like written in your manual Rx to Rx, Tx to Tx.  3,3v to 3,3v and GND to GND.
Now I swapped Rx to Tx on ESP and Tx to Rx as usual.

bbqkees, 2019-11-11:
Did you connect RX to D7 and TX to D8?

kkberg, 2019-11-11:
I'm so sorry, that's the reason. I now see 👍 
EMS Bus Connected with Rx active but Tx has been disabled (in listen only mode). Problem as always 50cm infront of the screen. Now I'm really excited to see data. I'll give you a message later.

kkberg, 2019-11-11:
| {"selFlowTemp":5,"selBurnPow":0,"curBurnPow":0,"pumpMod":0,"outdoorTemp":3.5,"wWCurTmp":51.9,"wWCurFlow":0,"curFlowTemp":27.2,"burnGas":"off","flameCurr":0,"heatPmp":"off","fanWork":"off","ignWork":"off","wWCirc":"off","wWHeat":"off","wWStarts":981,"wWWorkM":17319,"burnStarts":1989,"burnWorkMin":35777,"heatWorkMin":18458,"ServiceCode":"0H","ServiceCodeNumber":203}
-- | --
2019-11-11 09:40:48 | tapwater_active | 0
2019-11-11 09:40:48 | heating_active | 0

bbqkees, 2019-11-11:
Ok no problem. It's a common mistake.
I'll close the issue now.

closed 2019-11-11 by bbqkees

kkberg, 2019-11-11:
Thank you again, you made my day. Sorry for making trouble. You are now also mentioned with a link on Alex Kabza's webside. I'm also talking to him because of another interface! http://www.kabza.de/MyHome/EMSbus.html

bbqkees, 2019-11-11:
The reason D7 and D8 are used on the Wemos instead of the normal TX and RX is because at boot the ESP sends out some debug information on TX, thereby disturbing the EMS bus.
With the pinswap function the UART is set to the alternative pins D7 and D8 after this debug stuff is sent so the EMS bus is never disturbed.

kkberg, 2019-11-11:
Okay, I red it but didn't take care when connecting the ESP. Now I get data but what happenes is, that my RC310 goes on failure and is only shwoing a blinking screen. Any ideas?

kkberg, 2019-11-11:
After letting it work for some time, connection to EMS-Bus ist lost and one capacitor is getten very hot.
It's C1 on the pcb. What could happened to cause this?